pfalcon/pycopy #29: ure: mixing named classes and characters in a match set raises ValueError

c0d3z3r0, 2019-05-03:
Tested on `unix` and `esp8266`, master branch

~~~
>>> ure.compile('\d+')
<re 3fff02f0>

>>> ure.compile('[abc0-9]+')
<re 3fff0580>

>>> ure.compile('[abc\d]+')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Error in regex

>>> ure.compile('[\w\d]+')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Error in regex
~~~

pfalcon, 2019-05-03:
Yes, this was never supported (never parsed properly), With recent changes (WIP so far), any (well, most) unsupported regex syntax now raise an error instead of silent unexpected behavior.

pfalcon, 2019-05-07:
Ok, these changes are finished now. Nothing really new from when this issue was posted, they were just promoted from "WIP" to "finalized". The docs were updated yet then too: https://pycopy.readthedocs.io/en/latest/library/ure.html

closed 2019-05-07 by pfalcon