pfalcon/pycopy #30: uasyncio / EAGAIN

c0d3z3r0, 2019-05-30:
Hi there,

I am having the same issue that should have been fixed in https://github.com/micropython/micropython/pull/4322

Test code:

~~~
import uasyncio as asyncio

@asyncio.coroutine
async def serve(reader, writer):
  request = yield from reader.read()
  print((yield from reader.read()))
  yield from writer.awrite("ok\r\n")
  yield from writer.aclose()

def run():
  loop = asyncio.get_event_loop()
  loop.create_task(asyncio.start_server(serve, "127.0.0.1", 80))
  loop.run_forever()
  loop.close()
~~~

~~~
MicroPython v1.10-579-ga64fa0e8e on 2019-05-30; ESP module with ESP8266
Type "help()" for more information.
>>> import micropython
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 37952, used: 5280, free: 32672
 No. of 1-blocks: 24, 2-blocks: 11, max blk sz: 264, max free sz: 2025
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import uasyncio as asyncio
=== 
=== @asyncio.coroutine
=== async def serve(reader, writer):
===   request = yield from reader.read()
===   print((yield from reader.read()))
===   yield from writer.awrite("ok\r\n")
===   yield from writer.aclose()
=== 
=== def run():
===   loop = asyncio.get_event_loop()
===   loop.create_task(asyncio.start_server(serve, "127.0.0.1", 80))
===   loop.run_forever()
===   loop.close()
>>> run()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 13, in run
  File "uasyncio/core.py", line 163, in run_forever
  File "uasyncio/core.py", line 116, in run_forever
  File "uasyncio/__init__.py", line 252, in start_server
OSError: [Errno 11] EAGAIN
>>> run()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 13, in run
  File "uasyncio/core.py", line 163, in run_forever
  File "uasyncio/core.py", line 116, in run_forever
  File "uasyncio/__init__.py", line 245, in start_server
OSError: [Errno 12] ENOMEM
>>> run()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 13, in run
  File "uasyncio/core.py", line 163, in run_forever
  File "uasyncio/core.py", line 116, in run_forever
  File "uasyncio/__init__.py", line 245, in start_server
OSError: [Errno 12] ENOMEM
>>> micropython.mem_info()
stack: 2112 out of 8192
GC: total: 37952, used: 12112, free: 25840
 No. of 1-blocks: 152, 2-blocks: 35, max blk sz: 264, max free sz: 1605
~~~

What could be wrong here?

pfalcon, 2019-05-30:
I'm not sure. The errors are definitely coming for modlwip sockets. I'm rebasing on the micropython upstream, so that fix is included in my tree. I believe there were more fixes to modlwip recently, the latest revision in this repo is 4dcb7d093de4d, you may want to try build.

Unfortunately, I don't have time now to look into esp8266 issues.

c0d3z3r0, 2019-05-30:
Hm... interesting... I removed these two lines from boot.py to disable the AP mode - no more EAGAIN...

~~~
import network
network.WLAN(network.AP_IF).active(False)
~~~

That means the error happens only when not in AP mode, which currently makes the AP mandatory...
Connecting to a WiFi network does not help.

The ENOMEM error is still there, when doing Ctrl-C and restart run()...

c0d3z3r0, 2019-05-30:
> I believe there were more fixes to modlwip recently, the latest revision in this repo is 4dcb7d0, you may want to try build.

I don‘t get this. Your master branch is based on current upstream‘s master, so yes, this patch is included. What other patches do you mean?
I tested this with 4dcb7d0 - I haven’t tested with upstream, yet. I will do, when I‘m back home later this day

c0d3z3r0, 2019-05-30:
Tested with raw sockets, no asyncio... same problem:

~~~
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import socket as _socket
=== ai = _socket.getaddrinfo("127.0.0.1", 85, 0, _socket.SOCK_STREAM)
=== ai = ai[0]
=== s = _socket.socket(ai[0], ai[1], ai[2])
=== s.setblocking(False)
=== s.setsockopt(_socket.SOL_SOCKET, _socket.SO_REUSEADDR, 1)
=== s.bind(ai[-1])
=== s.listen(0)
=== 
>>> s2, client_addr = s.accept()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 11] EAGAIN
~~~

Catching EAGAIN works but this only circumvents that bug...

~~~
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import socket as _socket
=== ai = _socket.getaddrinfo("127.0.0.1", 85, 0, _socket.SOCK_STREAM)
=== ai = ai[0]
=== s = _socket.socket(ai[0], ai[1], ai[2])
=== s.setblocking(False)
=== s.setsockopt(_socket.SOL_SOCKET, _socket.SO_REUSEADDR, 1)
=== s.bind(ai[-1])
=== s.listen(0)
=== 
>>> try:
>>>   s2, client_addr = s.accept()
>>> except OSError as e:
>>>   if e.args[0] == uerrno.EAGAIN:
>>>     continue
>>>

~~~

closed 2019-05-31 by c0d3z3r0

c0d3z3r0, 2019-05-31:
https://github.com/pfalcon/micropython-lib/pull/33

JuniorJPDJ, 2019-08-08:
@c0d3z3r0 why you have closed this?

c0d3z3r0, 2019-08-11:
@JuniorJPDJ because this is an upstream issue ;)