pfalcon/pycopy #31: Unknown type name 'ppp_pcb'

jfdona23, 2019-06-04:
Compiling PyCopy 1.11 with the suggested ESP-IDF branch I get this error: **unknown type name 'ppp_pcb'**

Seems to be happening in Loboris fork too: https://github.com/loboris/ESP32-PPPOS-EXAMPLE/issues/15

So far, I can skip this error by adding this line to Makefile:

> INC += -I$(TOP)/lib/lwip/src/include

But now I've another error:

> In file included from network_ppp.c:39:0:
> ../../lib/lwip/src/include/lwip/sockets.h:428:2: error: #error LWIP_SOCKET_OFFSET does not work with external FD_SET!
> #error LWIP_SOCKET_OFFSET does not work with external FD_SET!

Fo the last error, I commented the line 408 in $(TOP)/lib/lwip/src/include/lwip/sockets.h

> /* FD_SET used for lwip_select */ // LINE 407 
> //#ifndef FD_SET  // LINE 408

...and added these in the next line (409):

> #ifdef LWIP_SOCKET_OFFSET
> #undef FD_SET
> #undef FD_CLR
> #undef FD_ISSET
> #undef FD_ZERO
> #undef _types_fd_set
> #undef fd_set

Source: https://github.com/espressif/esp-idf/issues/1141

But again, I got another error:
> ../../lib/lwip/src/include/lwip/arch.h:125:19: error: conflicting types for 'mem_ptr_t'
>  typedef uintptr_t mem_ptr_t;

All of this seems to be related to esp-idf since the change constantly. I'm working in the suggested branch for PyCopy 1.1: **5c88c5996**

Any help will be appreciated :)

👋

jfdona23, 2019-06-05:
Well well, I opted for delete my entire cloned repo and started over again. No issues this time!!
My best guess is maybe I missed something, or maybe the fact I was using GNUmakefile instead makefile. I'm running it in Microsoft WSL and in the rush I chose wrong my custome makefile.
Sorry @pfalcon for I've opened such useless thread 😞

closed 2019-06-05 by jfdona23

pfalcon, 2019-06-05:
Right, in my branch, I don't make any changes related to esp32. So, the situation would be exactly the same as upstream. My plan was to update to the latest upstream and tell you so ;-). Glad you didn't wait and tried to figure it on your side.

The cause also might be that you didn't update git submodules, you'd need to run `git submodule sync` / `git submodule update` regularly.