pfalcon/pycopy #39: upip fails

klauweg, 2019-10-10:
I've tried to install picoweb with upip. But it fails.
pycopy is version aa986f5eebe669
compiled with esp-idf v3.3 6ccb4cf5b7d1fdd
running on esp32 (doit-devkit)

```
Connection successful
('192.168.70.71', '255.255.255.0', '192.168.70.1', '192.168.200.1')
>>> import upip
>>> upip.install("picoweb")
Installing to: /lib/
Warning: pypi.org SSL certificate is not validated
Installing picoweb from https://files.pythonhosted.org/packages/a7/cb/1b48103a22f8d67985af5e30ee157c195d2a37b12864e6e8744ad8e7e1b6/picoweb-1.7.2.tar.gz
Error installing 'picoweb': TypeError('object with buffer protocol required',), packages may be partially installed
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "upip.py", line 248, in install
  File "upip.py", line 236, in install
  File "upip.py", line 209, in install_pkg
TypeError: object with buffer protocol required
>>>
```

pfalcon, 2019-10-10:
Thanks for the report. I see the problem. Please try the changes in the latest master. Be sure to update with `git pull --rebase`.

pfalcon, 2019-10-20:
Assuming the issue fixed.

closed 2019-10-20 by pfalcon