pfalcon/pycopy #40: running picoweb blocks timer

klauweg, 2019-10-10:
In the following screenshot you can see the signal of a timer toggling an io pin every 5ms.

like so:
```
tm = machine.Timer(0) 
tm.init( period=5, mode=tm.PERIODIC, callback=update)
```
furthermore there are dropouts of almost 50ms when picoweb is serving a http request in parallel as we can see in the middle of the Screenshot.

I know this behaviour from micropython/MicroWebSrv. I hoped of some improvement due to this asyncio stuff. But maybe this is totally normal. Or am i doing something wrong?

[image]

pfalcon, 2019-10-10:
Is this also ESP32, as you report in https://github.com/pfalcon/pycopy/issues/39 ?

So, reference platform for both Pycopy and picoweb is Unix port (and Linux, etc. systems). I don't do anything in picoweb which should affect stuff like machine.Timer. And if it's implemented properly (i.e. backed by real hardware timer), it should not be affected by Picoweb in particular, and network activity in general (and that's where I may guess, the problem lies).

But ways it's implemented in various MicroPython baremetal ports vary widely, and that's exactly why I concentrate on the Unix port for time being, as I don't have resources to deal with all that variation. The only thing I can say is that I don't make any changes to baremetal ports (again, for now), so hardware APIs should work the same way as in MicroPython. So, I'd suggest to report issues/concerns like above there.

klauweg, 2019-10-10:
> Is this also ESP32, as you report in #39 ?
Yes, that's correct.

I understand your further explanations. So i think you can close the issue.

closed 2019-10-11 by pfalcon