pfalcon/pycopy #42: CAN bus - not receiving frames of type extframe=False

JohnieBraaf, 2020-01-08:
There apprears to be an issue in the CAN drivers which prevents receiving messages with non extended id's.

As a test I'm using a pyboard-D to send data between the 2 CAN ports.
This works fine when the can filter is set to extframe=True.
`
can.init(mode=pyb.CAN.NORMAL, extframe=True, prescaler=4, sjw=1, bs1=7, bs2=1)
`

However, when changed to False, there are no incoming frames detected.
I can see the frames with a CAN bus analyzer, but the rx callback does not trigger.
`
can.init(mode=pyb.CAN.NORMAL, extframe=False, prescaler=4, sjw=1, bs1=7, bs2=1
`

I've tried sending frames with non-extended id's which works fine.
Any pointers into where this can be fixed? Is it STM32 specific?

JohnieBraaf, 2020-01-08:
I think I might have found where this goes wrong.   

I'm using CAN.MASK32 with a blank values like this:
`
can.setfilter(bank=self.itf-1, mode=pyb.CAN.MASK32, fifo=self.itf-1, params=(0x0, 0x0))
`

Now when I trace back the code, I come to pyb_can.c, where the filtermasks are set.
They appear to be designed for extended identifier format judging from the high value 0x1FFFE000.
 https://github.com/pfalcon/pycopy/blob/0b7e3d16c281e95c76fcec83f3e0349d99af4573/ports/stm32/pyb_can.c#L710-L713

Now when I change it to blank, I start receiving the packets.
`
filter.FilterIdHigh = 0x0000;
filter.FilterIdLow = 0x0000;
`
`
filter.FilterMaskIdHigh = 0x0000;
filter.FilterMaskIdLow  = 0x0000;
`

Somehow, the logic needs to be extended to give the correct bounds for non-extended frames.
With the above hack id's are not correct.

pfalcon, 2020-01-09:
Thanks for the report, but the Pycopy projects concentrates on the core languages, not on hardware-specific ports and facilities. Those come directly from the upstream MicroPython, so should be reported/fixed there.

JohnieBraaf, 2020-01-09:
thnx, I've raised the issue upstream. https://github.com/micropython/micropython/issues/5508

closed 2020-01-09 by JohnieBraaf