pfalcon/pycopy #48: Erro trying to compile pycopy stable v3.2.1

pronyxhack, 2020-06-25:
Parameters:

git clone https://github.com/pfalcon/pycopy.git
cd pycopy
git checkout 48337873c6acf7f3aaa69104b725de528fe2092f
git submodule update --init --recursive

$ cd ports/esp32
$ make ESPIDF
$ export ESPIDF=$HOME/Downloads/ESP32/esp-idf  # Or any path you like.
$ mkdir -p $ESPIDF
$ cd $ESPIDF
$ git clone https://github.com/espressif/esp-idf.git $ESPIDF
$ git checkout 4c81978a3e2220674a432a588292a4c860eef27b
$ git submodule update --init --recursive

CC ../../extmod/virtpin.c
CC ../../extmod/machine_mem.c
CC ../../extmod/machine_pinbase.c
CC ../../extmod/machine_signal.c
CC ../../extmod/machine_pulse.c
CC ../../extmod/machine_i2c.c
CC ../../extmod/machine_spi.c
CC ../../extmod/modbluetooth.c
CC ../../extmod/modussl_axtls.c
CC ../../extmod/modussl_mbedtls.c
../../extmod/modussl_mbedtls.c: In function 'socket_new':
../../extmod/modussl_mbedtls.c:252:5: error: implicit declaration of function 'mbedtls_ssl_endpoint'; did you mean 'mbedtls_ssl_conf_endpoint'? [-Werror=implicit-function-declaration]
     mbedtls_ssl_endpoint(&o->ssl, args->server_side.u_bool ? MBEDTLS_SSL_IS_SERVER : MBEDTLS_SSL_IS_CLIENT);
     ^~~~~~~~~~~~~~~~~~~~
     mbedtls_ssl_conf_endpoint
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:63: build-GENERIC/extmod/modussl_mbedtls.o] Error 1

pfalcon, 2020-06-26:
Pycopy requires special mbedTLS branch, see https://github.com/pfalcon/pycopy/blob/master/.travis.yml#L356 how to build esp32 with it.

motjebben, 2020-06-29:
I had this same problem as @pronyxhack and tried @pfalcon 's suggestion, but ran into different new problems.  I'll try cleaning the directories and building again...

pronyxhack, 2020-06-30:
> Pycopy requires special mbedTLS branch, see https://github.com/pfalcon/pycopy/blob/master/.travis.yml#L356 how to build esp32 with it.

The problem was solved by compiling with the indicated instructions. Thank you pfalcon!

closed 2020-06-30 by pronyxhack

pfalcon, 2020-07-01:
Glad that it worked. I'm not sure if this ticket should be left open, to help other folks, like @motjebben. Anyway, can be reopened if needed.

Otherwise, I'm committed to keep baremetal platforms from MicroPython building, but the main target of Pycopy is unix port, and the current direction is finishing it (then hopefully propagate changes back to baremetal ports). So, changes like are expected and necessary. Bottom line: if there's interest to run Pycopy on esp32, etc., then more people should participate in maintaining it. Otherwise, well, I just try to keep Travis from breaking for as long as I can ;-).

motjebben, 2020-07-01:
I'm set, too, thanks!  The next error I ran into appears to have to do with OS-X not liking deep linkages, and I'll work on that myself. Anyway, your pointer above was perfect!  Be well!