pfalcon/pycopy #49: ESP32 - Error when building firmware: implicit declaration of function 'mbedtls_ssl_endpoint'

AndyFug, 2020-07-10:
Hi,

I'm receiving this error while trying to build ESP32 firmware:

```
CC ../../extmod/modussl_mbedtls.c
../../extmod/modussl_mbedtls.c: In function 'socket_new':
../../extmod/modussl_mbedtls.c:253:5: error: implicit declaration of function 'mbedtls_ssl_endpoint' [-Werror=implicit-function-declaration]
     mbedtls_ssl_endpoint(&o->ssl, args->server_side.u_bool ? MBEDTLS_SSL_IS_SERVER : MBEDTLS_SSL_IS_CLIENT);
     ^
cc1: all warnings being treated as errors
../../py/mkrules.mk:63: recipe for target 'build-GENERIC/extmod/modussl_mbedtls.o' failed
make: *** [build-GENERIC/extmod/modussl_mbedtls.o] Error 1

```
I followed [this](https://github.com/pfalcon/pycopy/blob/master/ports/esp32/README.md) advice from the README to try to build the firmware, using ESP-IDF v3.3 (checking IDF compatibility with the hash etc as directed) and the ESP32 toolchain for Linux from [here](https://docs.espressif.com/projects/esp-idf/en/v3.3.2/get-started/linux-setup.html)  I think I've followed all the steps correctly.

**Additional details:**
- Pycopy hash: 0bdd88205
- ESP-IDF v3.3 hash: 9e70825d1
- ESP32 Toolchain download: https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz
- Linux version: Ubuntu Bionic 64
- Unix and ESP8266 ports are building without this error

Because I'm not very experienced at this, I'm not entirely sure what other information might be relevant to the issue.  Let me know if there is anything else I can provide to help.  I tried running make with --keep-going and --ignore, but it doesn't appear to be completing fully because application.bin is not being created (due to modussl_mbedtls.o missing).

Any help greatly appreciated.  I know these ports aren't a priority, but if it's something I'm doing wrong or it's a simple fix/easy work-around, then finding that solution would be much appreciated.  Apologies also if the issue has already been noted somewhere that I haven't seen.

Many thanks,

Andy

pfalcon, 2020-07-10:
Duplicate of https://github.com/pfalcon/pycopy/issues/48.

AndyFug, 2020-07-10:
Thanks.  Sorry I didn't see that... I should check over closed issues in future.

pfalcon, 2020-07-10:
I suspected that issue should be kept open... But let me reopen this instead, as it has more clear title.

gfrornermichi, 2021-01-09:
Hi, 
as you moved from travis the solution is not available anymore - would it be possible to repost the solution, the special mbed_tls version that is needed?
Many thanks
Michael

pfalcon, 2021-03-23:
Ok, there were no further reports of this, and CI have been green re: ESP32 port all this time, so let me close this.

closed 2021-03-23 by pfalcon