pfalcon/pycopy #51: RMT Neopixel support?

itdaniher, 2020-08-09:
When last working on larger-scale art projects involving Neopixel / WS2812 LEDs, I made use of [Lobo's micropython fork](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo) which uses [RMT for generating the digital waveform](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/components/micropython/esp32/machine_neopixel.c). Would an improved implementation of `esp_neopixel_write` be welcome here? Bitbanging seems like a bad use of CPU cycles when RMT is available.

pfalcon, 2020-08-09:
Thanks for the submission, but the Pycopy project concentrates on the *language*, not on hardware-specific features of particular implementations. To give some comparison, "RMT Neopixel" has the same connection to Pycopy as to CPython (https://www.python.org/). You may have better luck with the original MicroPython, which exactly concentrates on MCU side of things.

closed 2020-08-09 by pfalcon

itdaniher, 2020-08-09:
Thanks for your excellent work all the same!