pfalcon/pycopy #53: ure: Escaping does not work in character set

iBobik, 2020-09-21:
```
>>> import ure as re
>>> re.compile(r'\.')
>>> re.compile(r'[a\.]')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: error in regex
```

Pycopy v3.1.5

pfalcon, 2020-09-21:
Yes, Pycopy's `ure` is a minimalist regex module, in turn based on a minimalist regex library, https://github.com/pfalcon/re1.5 . The latter supports only the minimal required number of escapes. Any other escapes can be encoded on Python level instead.

In this particular case, it makes no sense to escape a dot in a set, it can (and for `ure`, should) be written as is. The only char which can be quoted in a set is `]` (for `-`, it just should be written very first in the set.)

closed 2020-09-21 by pfalcon

iBobik, 2020-09-21:
Thank you for a quick reply.

Aha, then it is good to mention this in docs.

E.g. after "Match set of characters. Individual characters and ranges are
supported, including negated sets (e.g. [^a-c]).“
add something like: "Only `]` escaping is supported here, other characters
are treated without their special meaning. If you need to enumerate `-`
character it just should be written very first in the set."

Sorry for my English :-)

pfalcon, 2020-09-21:
Good point, docs are updated: https://pycopy.readthedocs.io/en/latest/library/ure.html . Note that escaping behavior outside character sets was already described previously.

The whole behavior is similar to the recent versions of CPython, which error out on unknown escapes:

```
Python 3.8.5 (default, Jul 20 2020, 19:48:14) 
[GCC 7.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import re
>>> re.compile(r"\g")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
...
re.error: bad escape \g at position 0
```

Except that CPython does that only for escapes outside char sets, but Pycopy inside too ;-). And of course, it supports less escapes in the first place (but again, on any unsupported, it errors out).

iBobik, 2020-09-21:
Nice :-)

Btw, is it the same with Micropython?

Jan Pobořil