pfalcon/pycopy #54: how to identify pycopy versus micropython during runtime

Josverl, 2020-11-01:
Hi ,
I'm looking for a way to identify `pycopy` ( versus `micropython` ) while running a script / module.

if i look at the unix port, this information does not seem to be availalbe directly 
``` python 
>>> sys.implementation
(name='micropython', version=(3, 3, 2))
```
I could make the assumption that that a version > 2.11.0 is an indicator of the pycopy runtime/firmware,
but hope/assume that there would be another, more reliable, way to determine this.

Any suggestion for something that i could test for ?

pfalcon, 2020-11-01:
Why do you need to do that?

Josverl, 2020-11-01:
I'm running a tool  to generate stubs on the board
These then can be used for code completion in addition to the cpython stubs.

To store stubs for different firmwares and versions I use a folder structure using the  the port and version. 
I'm looking for add the 'firmware family' to provide distinction between micropython, pycopy, loboris, Pycom and others. 

It would be useful if that coukd be done by the micropython-stubber rather than through user input

pfalcon, 2020-11-01:
I see. I'm not familiar enough with that "stubs" business, and assume you're really need to distinguish various implementation, instead of being "data-driven", i.e. see what modules are available and what functions/classes in them.

So, for the time being, Pycopy is intended to be "MicroPython compatible", that's why `sys.implementation`
has `name='micropython'`, and I don't intend to change that for the time being (knowing that people use that to test for MicroPython).

Testing a version to differentiate the two is surely too hacky. I can propose to use availability of the `pycopy` module to check whether it's Pycopy. But: a) it's relatively recent introduction (but Pycopy is experimental project, and too-old versions are effectively not supported); b) as an implementation detail, it's currently an alias for `micropython` module; c) I believe it's implemented only in the unix port (and that's what actually maintained, microcontroller ports aren't primary target of Pycopy, at least for now).

Josverl, 2020-11-02:
Thanks Paul,

testing for the `pycopy` module seems logical.  
I'll give it a try on the unix port and run a spotcheck on some of the other platforms as well.

thanks for clarifying.

closed 2020-11-02 by Josverl