pfalcon/pycopy #55: build failed for android port

Lwenkun, 2021-03-16:
I use build-android.sh in "cross" branch to build but get some errors. Is there tutorial about how to build for android port on Macos? 
My steps:
```
make-standalone-toolchain.sh
cd ports/unix
make axtls
./build-android-sh
```
the last step throws error:
```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build-micropython-android/lib/libffi; cd build-micropython-android/lib/libffi; \
        /Users/liwenkun/Desktop/pycopy/lib/libffi/configure --host=arm-linux-androideabi --prefix=$PWD/out --disable-structs CC="clang" CXX="arm-linux-androideabi-g++" LD="arm-linux-androideabi-ld" CFLAGS="-Os -fomit-frame-pointer -fstrict-aliasing -ffast-math -fno-exceptions"; \
        /Applications/Xcode.app/Contents/Developer/usr/bin/make install-exec-recursive; /Applications/Xcode.app/Contents/Developer/usr/bin/make -C include install-data-am
checking build system type... x86_64-apple-darwin20.3.0
checking host system type... arm-unknown-linux-androideabi
checking target system type... arm-unknown-linux-androideabi
checking for gsed... sed
checking for a BSD-compatible install... /usr/local/bin/ginstall -c
checking whether build environment is sane... yes
checking for arm-linux-androideabi-strip... arm-linux-androideabi-strip
checking for a thread-safe mkdir -p... /usr/local/bin/gmkdir -p
checking for gawk... no
checking for mawk... no
checking for nawk... no
checking for awk... awk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for arm-linux-androideabi-gcc... clang
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether clang accepts -g... yes
checking for clang option to accept ISO C89... none needed
checking whether clang understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of clang... gcc3
checking whether we are using the GNU C++ compiler... yes
checking whether arm-linux-androideabi-g++ accepts -g... yes
checking dependency style of arm-linux-androideabi-g++... gcc3
checking dependency style of clang... gcc3
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by clang... arm-linux-androideabi-ld
checking if the linker (arm-linux-androideabi-ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /Users/liwenkun/android-sdk/ndk/16.1.4479499/toolchains/arm-linux-androideabi-4.9/prebuilt/darwin-x86_64/bin/arm-linux-androideabi-nm -B
checking the name lister (/Users/liwenkun/android-sdk/ndk/16.1.4479499/toolchains/arm-linux-androideabi-4.9/prebuilt/darwin-x86_64/bin/arm-linux-androideabi-nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 786432
checking how to convert x86_64-apple-darwin20.3.0 file names to arm-unknown-linux-androideabi format... func_convert_file_noop
checking how to convert x86_64-apple-darwin20.3.0 file names to toolchain format... func_convert_file_noop
checking for arm-linux-androideabi-ld option to reload object files... -r
checking for arm-linux-androideabi-objdump... arm-linux-androideabi-objdump
checking how to recognize dependent libraries... pass_all
checking for arm-linux-androideabi-dlltool... no
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for arm-linux-androideabi-ar... arm-linux-androideabi-ar
checking for archiver @FILE support... @
checking for arm-linux-androideabi-strip... (cached) arm-linux-androideabi-strip
checking for arm-linux-androideabi-ranlib... arm-linux-androideabi-ranlib
checking command to parse /Users/liwenkun/android-sdk/ndk/16.1.4479499/toolchains/arm-linux-androideabi-4.9/prebuilt/darwin-x86_64/bin/arm-linux-androideabi-nm -B output from clang object... failed
checking for sysroot... no
checking for a working dd... /bin/dd
[... 117 lines cut ...]
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I/Users/liwenkun/Desktop/pycopy/lib/libffi -I. -I/Users/liwenkun/Desktop/pycopy/lib/libffi/include -Iinclude -I/Users/liwenkun/Desktop/pycopy/lib/libffi/src -Os -fomit-frame-pointer -fstrict-aliasing -ffast-math -fno-exceptions -Wall -fexceptions -MT src/closures.lo -MD -MP -MF src/.deps/closures.Tpo -c /Users/liwenkun/Desktop/pycopy/lib/libffi/src/closures.c -o src/closures.o >/dev/null 2>&1
depbase=`echo src/arm/ffi.lo | sed 's|[^/]*$|.deps/&|;s|\.lo$||'`;\
        /bin/sh ./libtool  --tag=CC   --mode=compile clang -DHAVE_CONFIG_H -I. -I/Users/liwenkun/Desktop/pycopy/lib/libffi  -I. -I/Users/liwenkun/Desktop/pycopy/lib/libffi/include -Iinclude -I/Users/liwenkun/Desktop/pycopy/lib/libffi/src   -Os -fomit-frame-pointer -fstrict-aliasing -ffast-math -fno-exceptions  -Wall -fexceptions -MT src/arm/ffi.lo -MD -MP -MF $depbase.Tpo -c -o src/arm/ffi.lo /Users/liwenkun/Desktop/pycopy/lib/libffi/src/arm/ffi.c &&\
        mv -f $depbase.Tpo $depbase.Plo
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I/Users/liwenkun/Desktop/pycopy/lib/libffi -I. -I/Users/liwenkun/Desktop/pycopy/lib/libffi/include -Iinclude -I/Users/liwenkun/Desktop/pycopy/lib/libffi/src -Os -fomit-frame-pointer -fstrict-aliasing -ffast-math -fno-exceptions -Wall -fexceptions -MT src/arm/ffi.lo -MD -MP -MF src/arm/.deps/ffi.Tpo -c /Users/liwenkun/Desktop/pycopy/lib/libffi/src/arm/ffi.c  -fPIC -DPIC -o src/arm/.libs/ffi.o
/Users/liwenkun/Desktop/pycopy/lib/libffi/src/arm/ffi.c:804:3: error: implicit declaration of function '__clear_cache' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
  __clear_cache(closure->tramp, closure->tramp + 8);    /* clear data map */
  ^
1 error generated.
make[2]: *** [src/arm/ffi.lo] Error 1
make[1]: *** [install-exec-recursive] Error 1
 /usr/local/bin/gmkdir -p '/Users/liwenkun/Desktop/pycopy/ports/unix/build-micropython-android/lib/libffi/out/lib/libffi-3.99999/include'
 /usr/local/bin/ginstall -c -m 644 ffi.h ffitarget.h '/Users/liwenkun/Desktop/pycopy/ports/unix/build-micropython-android/lib/libffi/out/lib/libffi-3.99999/include'
mkdir -p build-micropython-android/genhdr
python ../../py/makeversionhdr.py build-micropython-android/genhdr/mpversion.h
GEN build-micropython-android/genhdr/mpversion.h
GEN build-micropython-android/genhdr/qstr.i.last
clang -E -DNO_QSTR -Ibuild-micropython-android/tmp -I../../lib/berkeley-db-1.xx/PORT/include -I. -I../.. -Ibuild-micropython-android -I../../lib/mp-readline -Wall -Werror -Wpointer-arith -Wuninitialized -std=gnu99 -DUNIX -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DMICROPY_PY_USSL=1 -DMICROPY_SSL_AXTLS=1 -I../../lib/axtls/ssl -I../../lib/axtls/crypto -I../../extmod/axtls-include -DMICROPY_PY_BTREE=1 -DMICROPY_USE_READLINE=1 -DMICROPY_PY_TERMIOS=1 -DMICROPY_PY_SOCKET=1 -DMICROPY_PY_THREAD=1 -DMICROPY_PY_THREAD_GIL=0 -Ibuild-micropython-android/lib/libffi/out/lib/libffi-3.99999/include -DMICROPY_PY_FFI=1 -I/usr/lib/jvm/java-7-openjdk-amd64/include -DMICROPY_PY_JNI=1 -Os -fdata-sections -ffunction-sections -DNDEBUG  -U _FORTIFY_SOURCE -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -DMPZ_DIG_SIZE=16  ../../lib/axtls/ssl/asn1.c ../../lib/axtls/ssl/loader.c ../../lib/axtls/ssl/tls1.c ../../lib/axtls/ssl/tls1_svr.c ../../lib/axtls/ssl/tls1_clnt.c ../../lib/axtls/ssl/x509.c ../../lib/axtls/crypto/aes.c ../../lib/axtls/crypto/bigint.c ../../lib/axtls/crypto/crypto_misc.c ../../lib/axtls/crypto/hmac.c ../../lib/axtls/crypto/md5.c ../../lib/axtls/crypto/rsa.c ../../lib/axtls/crypto/sha1.c ../../extmod/modbtree.c ../../lib/berkeley-db-1.xx/btree/bt_close.c ../../lib/berkeley-db-1.xx/btree/bt_conv.c ../../lib/berkeley-db-1.xx/btree/bt_debug.c ../../lib/berkeley-db-1.xx/btree/bt_delete.c ../../lib/berkeley-db-1.xx/btree/bt_get.c ../../lib/berkeley-db-1.xx/btree/bt_open.c ../../lib/berkeley-db-1.xx/btree/bt_overflow.c ../../lib/berkeley-db-1.xx/btree/bt_page.c ../../lib/berkeley-db-1.xx/btree/bt_put.c ../../lib/berkeley-db-1.xx/btree/bt_search.c ../../lib/berkeley-db-1.xx/btree/bt_seq.c ../../lib/berkeley-db-1.xx/btree/bt_split.c ../../lib/berkeley-db-1.xx/btree/bt_utils.c ../ [... cut ...] ime.c moduselect.c alloc.c coverage.c fatfs_port.c ../../lib/mp-readline/readline.c ../../lib/timeutils/timeutils.c ../../lib/oofatfs/ff.c ../../lib/oofatfs/option/unicode.c ../../py/mpconfig.h mpconfigport.h >build-micropython-android/genhdr/qstr.i.last;
modjni.c:36:10: fatal error: 'jni.h' file not found
#include <jni.h>
         ^~~~~~~
1 error generated.
make: *** [build-micropython-android/genhdr/qstr.i.last] Error 1
make: *** Deleting file `build-micropython-android/genhdr/qstr.i.last'
```

pfalcon, 2021-03-23:
No, there's no special tutorial AFAIK. The idea was that that script would capture basic steps needed to build it, but it definitely may be not complete. I myself didn't look into Android port for a while, and currently busy with other things in the project, but will try to look when time permits.

pfalcon, 2021-03-23:
Oh, btw, one things is that I always used Linux for it. (And I don't have access to MacOS.)