pfalcon/pycopy #59: Build fails (UNIX port)

hybroid, 2021-06-22:
Build log:
```
[hybroid@lip pycopy-3.6.0]$ make -C mpy-cross
make: Entering directory '/home/hybroid/dev/pycopy-3.6.0/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/moduledefs.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
mkdir -p build/lib/utils/
mkdir -p build/py/
CC ../py/mpstate.c
CC ../py/nlr.c
CC ../py/nlrx86.c
CC ../py/nlrx64.c
CC ../py/nlrthumb.c
CC ../py/nlraarch64.c
CC ../py/nlrpowerpc.c
CC ../py/nlrxtensa.c
CC ../py/nlrsetjmp.c
CC ../py/malloc.c
CC ../py/gc.c
CC ../py/pystack.c
CC ../py/qstr.c
CC ../py/vstr.c
CC ../py/mpprint.c
CC ../py/unicode.c
CC ../py/mpz.c
CC ../py/reader.c
CC ../py/lexer.c
CC ../py/parse.c
CC ../py/scope.c
CC ../py/compile.c
CC ../py/emitcommon.c
CC ../py/emitbc.c
CC ../py/asmbase.c
CC ../py/asmx64.c
CC ../py/emitnx64.c
CC ../py/asmx86.c
CC ../py/emitnx86.c
CC ../py/asmthumb.c
CC ../py/emitnthumb.c
CC ../py/emitinlinethumb.c
CC ../py/asmarm.c
CC ../py/emitnarm.c
CC ../py/asmxtensa.c
CC ../py/emitnxtensa.c
CC ../py/emitinlinextensa.c
CC ../py/emitnxtensawin.c
CC ../py/formatfloat.c
CC ../py/parsenumbase.c
CC ../py/parsenum.c
CC ../py/emitglue.c
CC ../py/persistentcode.c
CC ../py/runtime.c
CC ../py/runtime_utils.c
CC ../py/scheduler.c
CC ../py/strict_mode.c
CC ../py/nativeglue.c
CC ../py/pairheap.c
CC ../py/ringbuf.c
CC ../py/stackctrl.c
CC ../py/argcheck.c
CC ../py/warning.c
CC ../py/profile.c
CC ../py/map.c
CC ../py/obj.c
CC ../py/objarray.c
CC ../py/objattrtuple.c
CC ../py/objbool.c
CC ../py/objboundmeth.c
CC ../py/objbufreader.c
CC ../py/objcell.c
CC ../py/objclosure.c
CC ../py/objcomplex.c
CC ../py/objdeque.c
CC ../py/objdict.c
CC ../py/objenumerate.c
CC ../py/objexcept.c
CC ../py/objfilter.c
CC ../py/objfloat.c
CC ../py/objfun.c
CC ../py/objgenerator.c
CC ../py/objgetitemiter.c
CC ../py/objint.c
CC ../py/objint_longlong.c
CC ../py/objint_mpz.c
CC ../py/objlist.c
CC ../py/objmap.c
CC ../py/objmodule.c
CC ../py/objobject.c
CC ../py/objpolyiter.c
CC ../py/objproperty.c
CC ../py/objnone.c
CC ../py/objnamedtuple.c
CC ../py/objrange.c
CC ../py/objreversed.c
CC ../py/objroproxy.c
CC ../py/objset.c
CC ../py/objsingleton.c
CC ../py/objslice.c
CC ../py/objstr.c
CC ../py/objstrunicode.c
CC ../py/objstringio.c
CC ../py/objtextio.c
CC ../py/objtuple.c
CC ../py/objtype.c
CC ../py/objzip.c
CC ../py/opmethods.c
CC ../py/sequence.c
CC ../py/stream.c
CC ../py/binary.c
CC ../py/builtinimport.c
CC ../py/builtinimport.c
../py/builtinimport.c: In function 'mp_builtin___import__':
../py/builtinimport.c:513:36: error: 'imp.module_from_hook' may be used uninitialized [-Werror=maybe-uninitialized]
  513 |                         module_obj = imp.module_from_hook;
      |                         ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
../py/builtinimport.c:407:22: note: 'imp' declared here
  407 |     mp_import_info_t imp;
      |                      ^~~
cc1: all warnings being treated as errors
make: *** [../py/mkrules.mk:77: build/py/builtinimport.o] Error 1
make: Leaving directory '/home/hybroid/dev/pycopy-3.6.0/mpy-cross'
```
Used GCC:
```
[hybroid@lip pycopy-3.6.0]$ LC_ALL=C gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/11.1.0/lto-wrapper
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++,d --with-isl --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-install-libiberty --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-libunwind-exceptions --disable-werror gdc_include_dir=/usr/include/dlang/gdc
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.1.0 (GCC)
```

pfalcon, 2021-06-22:
Thanks for the report. That's apparently gcc11's caprice. I don't have one readily around, but will look into fixing that.

pfalcon, 2021-12-01:
This was fixed in the latest master. Note that GCC11 still fails to "make dev", due to imaginable warnings-as-errors in the upstream mbedTLS. General recommendation is to avoid overmodern technology like GCC11, and use tested and tried GCC5, 6, 7, 8, 9, or 10.

closed 2021-12-01 by pfalcon