pfalcon/pycopy #61: SSL related

joyhope, 2021-08-02:
The following code works in MicroPython but is unable to work in Pycopy. 

    def IsSSL(self) :
        return ( hasattr(ssl, 'SSLContext') and \
                 isinstance(self._socket, ssl.SSLSocket) )

Error infor:  AttributeError: 'module' object has no attribute 'SSLSocket'

How could make a code changed, so that the pycopy could work.

joyhope, 2021-08-05:
changed code 

            return ( hasattr(ssl, 'SSLContext') and \
                     isinstance(self._socket, ssl.SSLContext) )

It works. but how could i detected runtime environment pycopy or micropyhon.  The implementation.name is "micropython" in two environments.

pfalcon, 2021-08-05:
> but how could i detected runtime environment pycopy or micropyhon. The implementation.name is "micropython" in two environments.

The recommended way is by trying to import `pycopy` module, if it imports, that's Pycopy. See https://github.com/pfalcon/pycopy/issues/54

pfalcon, 2021-08-05:
But note that generally, you should just test for whatever attributes you really need/use, and Pycopy is the minimalist Python implementation, which strives to provide minimal number of methods required to cover most real-world useful usecases (at least in `ussl` module, there's `ssl` module in the pycopy-lib, which can be made as compatible with CPython as realistically possible).

joyhope, 2021-08-05:
Understand your suggestion.   Here is my problem backgroud.  Some open-source project is designed for MicroPython， but I prefer to use pycopy. The pycopy design goal completely matches my goal.  So I try to make the project works with pycopy.

pfalcon, 2021-08-07:
> but I prefer to use pycopy. The pycopy design goal completely matches my goal.

I'm glad to hear, thanks!

> Some open-source project is designed for MicroPython

But that's too unspecific, sorry. A generic answer would be "you would need to port it to Pycopy". In which way, I don't know based on the information provided. And indeed, you would know better anyway - you know what's the project you have at hand, you want to it does, you know how it should behave, you can test it against Pycopy, using the documentation https://pycopy.readthedocs.io/en/latest/library/ussl.html or actual Pycopy executable.

Then if you performed the porting work, definitely feel free to share the info, that would be appreciated. I from my side last half year working on porting bigger modules from CPython's stdlib. I for example just starting to land the ported "subprocess" module to pycopy-lib (WIP!): https://github.com/pfalcon/pycopy-lib/tree/master/subprocess

closed 2021-11-06 by pfalcon