pfalcon/pycopy #65: |Question| I cant install package pycopy-lib on Python 3.7.0 (Windows).

F0rerunner1997, 2022-01-20:
Hi guys I have problem with installation of pycopy-lib package into windows version of python 3.7.0 .

 I always get error message: 
,,c:\python\python37\python.exe -m pip install --no-cache-dir --user pycopy-lib --disable-pip-version-check

Collecting pycopy-lib
  Downloading pycopy-lib-0.2.2.tar.gz (768 bytes)
ERROR: pycopy-lib from https://files.pythonhosted.org/packages/4a/0e/79682be709fae5dee8dfae20fa7782707f349931eafdeada0cfd55a46db9/pycopy-lib-0.2.2.tar.gz#sha256=df1ac81085dd643e8db2326150098abdd58f1d8b78c1e74b5e39d9889161c6e5 does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.
Process returned with code 1,,

Any help greatly appreciated.

pfalcon, 2022-01-21:
> Hi guys I have problem with installation of pycopy-lib package into windows version of python 3.7.0 .

Why do you want to install pycopy-lib package into windows version of python 3.7.0 ?

F0rerunner1997, 2022-01-28:
Sorry but I dont understand your question? 
For which operating system is pycopy-lib ? (windows or linux?)
Or how can i use it/ install it?

Thank you for your answer.

pfalcon, 2022-01-28:
Please see the project page, https://github.com/pfalcon/pycopy-lib , for the information about pycopy-lib.

closed 2022-01-28 by pfalcon